bouffalolab/bl_iot_sdk #11: Cant flash BL602 due to `UnicodeDecodeError` in Flash Tool

berkutta, 2020-11-06:
Tried this on different Computers and with different DT-BL10 Boards. Pressed D8 and afterwards EN to enter Programming Mode before pressing Button in Flash Tool.

[image]

YafeiJin, 2020-11-07:
@berkutta Sorry for this delay. Cause it's hard for us to reproduce this issue. So I upload one trial version, please help us verify if it's workable. Thank you very much!

https://pan.baidu.com/s/1Uwv_6F8_s4gwpKHjtVmHag 提取码: hx4x

berkutta, 2020-11-07:
@YafeiJin Thanks, can you upload it to a different service than pan.baidu.com? I can't download from pan.baidu.com

YafeiJin, 2020-11-07:
@berkutta Your email?

berkutta, 2020-11-07:
@YafeiJin info@benjaminmarty.ch

berkutta, 2020-11-07:
Version v1.4.5.1 works for me. I uploaded the file I got from @YafeiJin to mega.nz so others can download it easily: https://mega.nz/file/h9U0HZoZ#JZdBFhXwwLcuVAW0uK91TRav2iP2dagKSlLD22jE_9g

Please also update the Repository so others won't face this issue.

closed 2020-11-07 by berkutta

YafeiJin, 2020-11-08:
Thanks for your help.
We will update this after some regular testing.