bouffalolab/bl_iot_sdk #16: Please provide Upload tool for Linux

oeidfhlsk, 2020-11-15:
Hi there,
I already got some DEV-Boards but cant seem to find an upload tool for linux machines. I cant even get  BlDevCube.exe from this repo running in a Win10 VM, because it seems this is not the right file taken from Pythoninstaller. Another version of BlDevCube i could find on the Internet is abuout 10 times bigger filesize-wise and works flawless. However, i would like to upload from native linux terminal.

[image]

YafeiJin, 2020-11-15:
Could you please provide the error info in Linux? os info also needed, thanks

oeidfhlsk, 2020-11-15:
Linux upload tool is simply missing from tools folder. Or am I missing something?

gamelaster, 2020-11-15:
Hi @oeidfhlsk , there you go:
https://pine64.gami.ee/bl602/BLFlashEnv

YafeiJin, 2020-11-15:
Really sorry for missing uploading the linux version, and We will correct soon.
@gamelaster Thanks.

oeidfhlsk, 2020-11-15:
@gamelaster Thanks. I could flash upload now for the first time.

gamelaster, 2020-11-15:
@oeidfhlsk probably dumb question, but do you have installed Python 3.7?

oeidfhlsk, 2020-11-15:
I did, but i just came up with the nex missing dll. Probably this is the solution, upload 

> output from dist/{filename}, rather then build/{filename}
(https://stackoverflow.com/questions/19225132/pyinstaller-not-working-on-simple-helloworld-program)

YafeiJin, 2020-11-21:
@oeidfhlsk I will close this issue for the open source tools is ready now. https://github.com/bouffalolab/BLOpenFlasher

closed 2020-11-21 by YafeiJin