bouffalolab/bl_iot_sdk #17: Bootloader protocol

renzenicolai, 2020-11-18:
Would it be possible for you to release information on the protocol used to program the BL602 over it's UART?

Additionally I'd appreciate it if you would consider releasing the source code of those parts of the bootrom which you are able to release.

This would allow the community to create easy to use flashing tools and boost integration of your ecosystem with existing development environments (like for example Arduino). Providing developers with an easy means to get started with your products would cause popularity to rise significantly, resulting in you getting noticed by potential customers.

Thank you.

YafeiJin, 2020-11-19:
Hi Renze,
Thanks. It is okey for document.
Actually I am writing one golang version which will be open source, but be delayed. It can be open if you think it is proper. 
The following is the latest status:
Generate Partition bin: done;
Combine boot info and compiled bin: finished, not verified;
Generate dtb bin: I don’t find the proper package, maybe provide python version;
Program protocol: done
The golang version was supposed only for mass production.

YafeiJin, 2020-11-21:
@renzenicolai Please refer to https://github.com/bouffalolab/BLOpenFlasher. It works, but still need to improve a lot. The protocol will be added soon

YafeiJin, 2020-11-22:
The doc was uploaded to https://github.com/bouffalolab/bl_docs/tree/main/BL602_ISP

closed 2020-11-22 by YafeiJin

renzenicolai, 2020-11-22:
Thank you very much @YafeiJin , the document is exactly what I was looking for!