bouffalolab/bl_iot_sdk #18: Recent commit broke build

robertlipe, 2020-11-25:
git checkout e231762bbb7f9678c072ce5b3483ef819977afcc (which, ironically, is "mine")
compiles and links customer_app/sdk_app_gpio successfully for me. (The build breaks trying to run flash_bin on MacOS because that's a Linux file and we've covered this in https://github.com/pine64/bl_iot_sdk/issues/25

If I follow that with:
% git checkout master
Previous HEAD position was e231762b Merge pull request #14 from robertlipe/clean_components
Switched to branch 'master'
Your branch is up to date with 'origin/master'.

make -j10 V=1

The link now fails:

/Users/robertlipe/src/buf_labs/bl_iot_sdk/toolchain/riscv/Darwin/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /Users/robertlipe/src/buf_labs/bl_iot_sdk/customer_app/sdk_app_gpio/build_out/bl602/libbl602.a(start.o): in function `.L0 ':
/Users/robertlipe/src/buf_labs/bl_iot_sdk/components/bl602/bl602/evb/src/boot/gcc/start.S:102: undefined reference to `__preinit_array_start'
/Users/robertlipe/src/buf_labs/bl_iot_sdk/toolchain/riscv/Darwin/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /Users/robertlipe/src/buf_labs/bl_iot_sdk/components/bl602/bl602/evb/src/boot/gcc/start.S:114: undefined reference to `__preinit_array_end'
/Users/robertlipe/src/buf_labs/bl_iot_sdk/toolchain/riscv/Darwin/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /Users/robertlipe/src/buf_labs/bl_iot_sdk/components/bl602/bl602/evb/src/boot/gcc/start.S:124: undefined reference to `__init_array_start'
/Users/robertlipe/src/buf_labs/bl_iot_sdk/toolchain/riscv/Darwin/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /Users/robertlipe/src/buf_labs/bl_iot_sdk/components/bl602/bl602/evb/src/boot/gcc/start.S:128: undefined reference to `__init_array_end'
collect2: error: ld returned 1 exit status
make: *** [/Users/robertlipe/src/buf_labs/bl_iot_sdk/customer_app/sdk_app_gpio/build_out/sdk_app_gpio.elf] Error 1

This is certainly a build break by @qwang s checkin at https://github.com/bouffalolab/bl_iot_sdk/commit/25fb041d1c9059843a14c6efe91225c49748a86b
That changed flash_rom.ld, but didn't change 
../../components//bl602/bl602/evb/ld/ram.ld
../../components//bl602/bl602/evb/ld/flash.ld
and probably should because start.S is now referencing those symbols

Yeah, this directory is being built 
... /bl602/bl602/evb/ld -T flash.ld -lbl602  -L/Users

@qwang, please roll this change back or deliver an updated version that includes this change to all the startups and builds correctly.

Thank you.

robertlipe, 2020-11-25:
I'm tired of fighting git/github to convert this into a pull request, but the fix will look something like:

commit ca3734303ffbf6ceac2bc62d4fd93ba0f1d1daab (HEAD -> build_fix)
Author: Robert Lipe <robertlipe@gmail.com>
Date:   Wed Nov 25 17:10:54 2020 -0600

    Attempted fix for broken build. Provide symbols for c++ sections now called at startup.

    Build was broken by https://github.com/bouffalolab/bl_iot_sdk/commit/25fb041d1c9059843a14c6efe91225c49748a86b#diff-23f8d7dfb4337009032bd4ac6327eeabdd070b785e960323a07716eded80478e
    and analyzed at
    https://github.com/bouffalolab/bl_iot_sdk/issues/18
    See https://github.com/pine64/bl_iot_sdk/issues/81#issuecomment-733976502

    Tested: app_gpio and app_i2c now successfully link. Please test something that runs ram.ld
    before committing.

    P.S. This project *needs* automated testing for proposed changes so developers aren't chasing build
    breaks.

diff --git a/components/bl602/bl602/evb/ld/flash.ld b/components/bl602/bl602/evb/ld/flash.ld
index a84b17fd..cf544c7d 100644
--- a/components/bl602/bl602/evb/ld/flash.ld
+++ b/components/bl602/bl602/evb/ld/flash.ld
@@ -68,6 +68,23 @@ SECTIONS
     *(.gnu.linkonce.r.*)
   } > flash
 
+  .preinit_array :
+  {
+    . = ALIGN(4);
+    __preinit_array_start = .;
+    KEEP (*(.preinit_array))
+    __preinit_array_end = .;
+  } > flash
+
+  .init_array :
+  {
+    . = ALIGN(4);
+    __init_array_start = .;
+    KEEP (*(SORT_BY_INIT_PRIORITY(.init_array.*)))
+    KEEP (*(.init_array))
+    __init_array_end = .;
+  } > flash
+
   /*put wifibss in the first place*/
   .wifibss         (NOLOAD) :
   {
diff --git a/components/bl602/bl602/evb/ld/flash_rom.ld b/components/bl602/bl602/evb/ld/flash_rom.ld
index 93f6ad7c..8a2b1201 100644
--- a/components/bl602/bl602/evb/ld/flash_rom.ld
+++ b/components/bl602/bl602/evb/ld/flash_rom.ld
@@ -97,6 +97,23 @@ SECTIONS
     __init_array_end = .;
   } > flash
 
+  .preinit_array :
+  {
+    . = ALIGN(4);
+    __preinit_array_start = .;
+    KEEP (*(.preinit_array))
+    __preinit_array_end = .;
+  } > flash
+
+  .init_array :
+  {
+    . = ALIGN(4);
+    __init_array_start = .;
+    KEEP (*(SORT_BY_INIT_PRIORITY(.init_array.*)))
+    KEEP (*(.init_array))
+    __init_array_end = .;
+  } > flash
+
   /*put wifibss in the first place*/
   .wifibss         (NOLOAD) :
   {
diff --git a/components/bl602/bl602/evb/ld/ram.ld b/components/bl602/bl602/evb/ld/ram.ld
index 34530641..c810b05a 100644
--- a/components/bl602/bl602/evb/ld/ram.ld
+++ b/components/bl602/bl602/evb/ld/ram.ld
@@ -73,6 +73,23 @@ SECTIONS
     *(COMMON)
   } >ram
 
+  .preinit_array :
+  {
+    . = ALIGN(4);
+    __preinit_array_start = .;
+    KEEP (*(.preinit_array))
+    __preinit_array_end = .;
+  } > ram
+
+  .init_array :
+  {
+    . = ALIGN(4);
+    __init_array_start = .;
+    KEEP (*(SORT_BY_INIT_PRIORITY(.init_array.*)))
+    KEEP (*(.init_array))
+    __init_array_end = .;
+  } > ram
+
   PROVIDE( _heap_start = . );
   PROVIDE( _heap_size = ADDR(.stack) - _heap_start );

YafeiJin, 2020-11-26:
Hi robert
Thanks for this fix. If you have trouble to generate one pull request, please upload the patch file. I would merge it manually. 
ram.ld and flash.ld were used only during bring-up chip, and currently flash_rom.ld is supposed to use by these samples.

robertlipe, 2020-11-26:
Thank you for your attention.

Perhaps the better fix with that information, then, is to drop the other
two and replace the use of -Tflash.ld with -Tflash_rom. I'm still awaiting
hardware, so I can't really say. I can mechanically do it if that's better
than you doing it and running whatever appropriate tests.

Here's the patch. There's no rocket surgery here: is't the original patch
basically copy/pasted in one case and tweaked to write the const RODATA to
RAM instead of FLASH in the other.