bouffalolab/bl_iot_sdk #21: I2C is not waiting to END when NACK happens

gamelaster, 2020-12-03:
Related to [this post](https://bbs.bouffalolab.com/d/6-i2c-is-busy-after-end-interrupt) on BBS, I found the reason why this happens.
[Here when NACK happens](https://github.com/bouffalolab/bl_iot_sdk/blob/e231762bbb7f9678c072ce5b3483ef819977afcc/components/hal_drv/bl602_hal/hal_i2c.c#L113), it calls `i2c_callback` which disables all interrupts and generally the I2C. Anyway, at that time, the I2C BUSY flag is still true, and there is not possible to immediately make another I2C transmission.

When we don't disable I2C on NACK, the flow is following:
I2C_TXF_INT > I2C_NAK_INT > I2C_NAK_INT > I2C_END_INT (and now is BUSY flag false).
So my solution is after NACK happen, don't disable interrupts or I2C, just set status and wait for END interrupt.
Then everything works like charm.

- gamiee

YafeiJin, 2020-12-03:
@gamelaster Thanks, we will fix this issue soon

gamelaster, 2020-12-03:
Also, don't use [i2c_set_freq](https://github.com/bouffalolab/bl_iot_sdk/blob/master/components/hal_drv/bl602_hal/bl_i2c.c#L45) function, use [I2C_ClockSet](https://github.com/bouffalolab/bl_iot_sdk/blob/master/components/bl602/bl602_std/bl602_std/StdDriver/Src/bl602_i2c.c#L356) instead, since `i2c_set_freq` don't set all clocks.

YafeiJin, 2020-12-04:
@gamelaster Please provide your sample code to reproduce this issue. Cause there are different levels of API in SDK. Thanks

gamelaster, 2020-12-04:
@YafeiJin the `i2c_set_freq` API doesn't work wrongly, it correctly sets the registers, but not all of them (and also, it doesn't check for current hclk freq). It doesn't use `GLB_Set_I2C_CLK`, so it's not possible to get higher I2C frequencies.
Example:
`i2c_set_freq(0, 100); // 100 KHz`
[image]
(As you can see, the clock frequency is barely 1000 Hz)
But when using I2C_ClockSet:
`I2C_ClockSet(0, 100000) `
[image]
(Now, there is everything OK)

YafeiJin, 2020-12-04:
[image]
Please try again by switching params

gamelaster, 2020-12-04:
@YafeiJin 
Ah yes, I'm blind 😁 thank you