bouffalolab/bl_iot_sdk #22: OpenOCD.exe miss depedencies

gamelaster, 2020-12-08:
Hi,

I tried to use OpenOCD.exe in Flasher, sadly, it miss following libraries: `libftdi1.dll`, `libusb-0-1-4.dll` and `libusb1.0.dll`.

I tried latest OpenOCD, it actually works, but it shows "WaitCmd" is missing, so maybe yours OpenOCD is modified?

YafeiJin, 2020-12-08:
What's your debugger device? Please refer to https://github.com/bouffalolab/bl_docs/tree/main/BL602_Openocd%26GDB
OpenOCD is the standard version.

gamelaster, 2020-12-08:
Hi Yafei,

The JTAG works OK for me, just [openocd.exe](https://github.com/bouffalolab/bl_iot_sdk/blob/master/tools/flash_tool/openocd.exe) in this repository don't work. (it miss following libraries: libftdi1.dll, libusb-0-1-4.dll and libusb1.0.dll). Else it works good, flashing through BLCubeEnv.exe but also debugging.

YafeiJin, 2020-12-09:
Uploaded these libraries:  libftdi1.dll, libusb-0-1-4.dll and libusb1.0.dll.

closed 2020-12-09 by YafeiJin