bouffalolab/bl_iot_sdk #23: bl60x_simple_flasher.py

berkutta, 2020-12-14:
Is this file available somewhere? Or how should development on Linux work otherwise?
```
# make CONFIG_CHIP_NAME=BL602 CONFIG_LINK_ROM=1 -j flash

****** Please SET BL60X_SDK_PATH ******
****** Trying SDK PATH [/home/user/prog/bl_iot_sdk/customer_app/bl602_demo_wifi/../..]
use exsting version.txt file
****** Please SET BL60X_SDK_PATH ******
****** Trying SDK PATH [/home/user/prog/bl_iot_sdk/customer_app/bl602_demo_wifi/../..]
use exsting version.txt file
AS build_out/bl602/evb/src/boot/gcc/entry.o
CC build_out/bl602/evb/src/debug.o
AS build_out/bl602/evb/src/boot/gcc/start.o
[...] 
Generating BIN File to /home/user/prog/bl_iot_sdk/customer_app/bl602_demo_wifi/build_out/whole_dts40M_pt2M_boot2release_ef7015.bin
Building Finish. To flash build output.
python3: can't open file '/home/user/prog/bl_iot_sdk/tools/flash_tool/core/bl60x_simple_flasher.py': [Errno 2] No such file or directory
make: *** [/home/user/prog/bl_iot_sdk/customer_app/bl602_demo_wifi/../../make_scripts_riscv/project.mk:523: flash] Error 2
```

YafeiJin, 2020-12-14:
@berkutta Please use "Bouffalo Lab Dev Cube For Ubuntu" from https://dev.bouffalolab.com/download

Discordia42, 2021-01-22:
OH FUCK YOU im not making no garbage account for your stupid website.
I know microsoft isnt better in this regard but open source is supposed to be actually open.

YafeiJin, 2021-01-23:
@Discordia42 If you need the open source version, please refer to https://github.com/bouffalolab/BLOpenFlasher
Also there are some other better options. Sorry for this inconvenience, cause our resources are occupied by new projects and new chips.
Python version:
https://github.com/stschake/bl60x-flash
https://github.com/renzenicolai/bl602tool
Rust version:
https://github.com/spacemeowx2/blflash
https://github.com/mkroman/bouffalo-cli

Discordia42, 2021-01-23:
@YafeiJin Thanks, i also figured that out later on, just have a bit of an allergic reaction to getting prompted to make an account.

closed 2021-02-01 by YafeiJin