bouffalolab/bl_iot_sdk #25: Bl602 ito sdk 中可以设置的频率宽度为什么和文档中的不一致?

ggggxiaolong, 2020-12-19:
https://github.com/bouffalolab/bl_iot_sdk/blob/3a6fef76a20409cd04227d81f4b4c27b054490f9/components/hal_drv/bl602_hal/hal_pwm.c#L475

文档: https://github.com/bouffalolab/bl_docs/tree/main/BL602_RM/zh_CN pdf:154页 支持的占空比 (n 为整数, 且 2 <= n <= 65535^2)

另外能不能添加一个控制舵机的pwm demo

gamelaster, 2020-12-19:
https://github.com/pine64/ArduinoCore-bouffalo/blob/main/cores/bl602/bl602-hal-pwm.c

ggggxiaolong, 2020-12-19:
> https://github.com/pine64/ArduinoCore-bouffalo/blob/main/cores/bl602/bl602-hal-pwm.c

Thanks, could you tell me how to set **pwm** 50Hz `frequency`,I'm a newer to code hardware , i want do drive servo, but  the source code `frequency` mush in 20k~80M.

gamelaster, 2020-12-19:
Hi @ggggxiaolong , just use `bl_pwm_set_freq(pwm_channel_id, 50);` and specify which PWM channel you use 😊 .

YafeiJin, 2020-12-19:
@ggggxiaolong There are two CLK sources for PWM, BusCLK(CPUCLK/2=BusCLK, 80M) and Xtal(40M), then PWM CLK value is got by PWM CLK Source/PWM DIV. PWM CLK is the basic unit to generate PWM signal. Period/Thre1/Thre2 are 
just counters, with which the final PWM signal will be generated. All limits lie in the register width instead of the SW code, that is used for ensuring the duty precision with 1%. I don't have HW on hand now, will provide sample code Next Monday.

ggggxiaolong, 2020-12-20:
@gamelaster  Thanks

ggggxiaolong, 2020-12-20:
@YafeiJin  when `LED_PWM_PERIOD=1000` it works,but when i set `LED_PWM_PERIOD=50` it not works, it was 2.984kHz,and when duty less than 3 **freq** will change
[demo code](https://github.com/SmartArduino/Doiting_BL/blob/master/examples/PWM/PWM/app_main.c)
```c
#define LED 1        //GPIO引脚定义
#define LED_CHANLE 1 //GPIO1所对应的PWM通道
#define LED_PWM_PERIOD 1000
bool value = false;

void PWM(void *arg)
{

    // hal_pwm_init(LED_CHANLE, LED); //PWM初始化
    // hal_pwm_freq_update(LED_CHANLE, LED_PWM_PERIOD);
    bl_pwm_set_freq(LED_CHANLE, LED_PWM_PERIOD);
    bl_pwm_set_duty(LED_CHANLE, 0);

    int i = 0;
    while (1)
    { //呼吸灯效果实现

        for (i = 20; i > -1; i--)
        {
            // hal_pwm_duty_set(LED_CHANLE, i * 100, 0);
            bl_pwm_set_duty(LED_CHANLE, i);
            vTaskDelay(400 / portTICK_RATE_MS); //改变延时时间可改变呼吸速度的快慢
        }
        for (i = 0; i < 21; i++)
        {
            // hal_pwm_duty_set(LED_CHANLE, i * 100, 0);
            bl_pwm_set_duty(LED_CHANLE, i);
            vTaskDelay(400 / portTICK_RATE_MS);
        }
        vTaskDelay(300 / portTICK_RATE_MS);
        LOGI(TAG, "PWM Recycle");
    }
}

void user_main(void)
{

    if (xTaskCreate(PWM, "PWM", 512, NULL, 6, NULL) != pdPASS) //创建PWM函数任务
    {
        LOGI(TAG, "xTaskCreate(GPIO) failed");
    }
}
```

YafeiJin, 2020-12-21:
@ggggxiaolong Temporally, Please use this patch to support low frequency PWM, which add one divClk parameter. With the sample of sdk_app_pwm, input the command, pwm_test, then measure GPIO2 to check. Thanks  
[Add_PWM_CLK_DIV.patch.zip](https://github.com/bouffalolab/bl_iot_sdk/files/5724173/Add_PWM_CLK_DIV.patch.zip)

ggggxiaolong, 2020-12-21:
@YafeiJin  not works, 0HZ-0%duty

YafeiJin, 2020-12-21:
@ggggxiaolong If possible, please provide more details on this issue. Thank a lot. Also, I uploaded the bin that I tested. PWM can output 50Hz, and duty could be changed.  
[sdk_app_pwm.bin.zip](https://github.com/bouffalolab/bl_iot_sdk/files/5725029/sdk_app_pwm.bin.zip)

ggggxiaolong, 2020-12-22:
@YafeiJin Ok, I will test provide bin file tonight

ggggxiaolong, 2020-12-22:
```
�Starting bl602 now....
Booting BL602 Chip...
██████╗ ██╗      ██████╗  ██████╗ ██████╗
██╔══██╗██║     ██╔════╝ ██╔═████╗╚════██╗
██████╔╝██║     ███████╗ ██║██╔██║ █████╔╝
██╔══██╗██║     ██╔═══██╗████╔╝██║██╔═══╝
██████╔╝███████╗╚██████╔╝╚██████╔╝███████╗
╚═════╝ ╚══════╝ ╚═════╝  ╚═════╝ ╚══════╝

------------------------------------------------------------
RISC-V Core Feature:RV32-ACFIMX
Build Version: release_bl_iot_sdk_1.6.11-1-g66bb28da-dirty
Build Version: release_bl_iot_sdk_1.6.11-1-g66bb28da-dirty
PHY   Version: a0_final-44-geb7fadd
RF    Version: f76e39a
Build Date: Dec 22 2020
Build Time: 20:26:23
------------------------------------------------------------
Heap 185772@0x42014254, 14624@0x420486e0

blog init set power on level 2, 2, 2.
[IRQ] Clearing and Disable all the pending IRQ...
[         0][INFO  : hal_boot2.c: 276] [HAL] [BOOT2] Active Partition[0] consumed 596 Bytes
======= PtTable_Config @0x4200d3c4=======
magicCode 0x54504642; version 0x0000; entryCnt 7; age 0; crc32 0x12DF9A26
idx  type device activeIndex     name   Address[0]  Address[1]  Length[0]   Length[1]   age
[00]  00     0         0            FW  0x00010000  0x000d8000  0x000c8000  0x00088000  0
[01]  02     0         0           mfg  0x00160000  0x00000000  0x00032000  0x00000000  0
[02]  03     0         0         media  0x00192000  0x00000000  0x00057000  0x00000000  0
[03]  04     0         0           PSM  0x001e9000  0x00000000  0x00008000  0x00000000  0
[04]  05     0         0           KEY  0x001f1000  0x00000000  0x00002000  0x00000000  0
[05]  06     0         0          DATA  0x001f3000  0x00000000  0x00005000  0x00000000  0
[06]  07     0         0       factory  0x001f8000  0x00000000  0x00007000  0x00000000  0
[         0][INFO  : bl_flash.c:  96] ======= FlashCfg magiccode @0x42049c18, code 0x47464346 =======
[         0][INFO  : bl_flash.c:  97]mid 		0xEF
[         0][INFO  : bl_flash.c:  98] clkDelay 	0x1
[         0][INFO  [0m: bl_flash.c:  99] clkInvert 	0x1
[         0][INFO  : bl_flash.c: 100] sector size	4KBytes
[         0][INFO  : bl_flash.c: 101] page size	256Bytes
[         0][INFO  : bl_flash.c: 102] ---------------------------------------------------------------
[         0][INFO  : hal_board.c: 857] [MAIN] [BOARD] [FLASH] addr from partition is 001f8000, ret is 0
[         0][INFO  : hal_board.c: 865] [MAIN] [BOARD] [XIP] addr from partition is 231e7000, ret is 0
[         0][INFO  : hal_board.c: 315] xtal_mode length 2
[         0][INFO  : hal_board.c: 319] xtal_mode is MF
[         0][INFO  : hal_board.c: 243] xtal dtb in DEC :36 36 0 60 60
[         0][INFO  : hal_board.c: 590] channel_div_table :
[         0][BUF: hal_board.c: 591] c4 4e ec 1e cb b7 fc 1e d2 20 0d 1f d8 89 1d 1f df f2 2d 1f e5 5b 3e 1f ec c4 4e 1f f2 2d 5f 1f f9 96 6f 1f 00 00 80 1f 06 69 90 1f 0d d2 a0 1f 13 3b 
[         0][BUF: hal_board.c: 591] b1 1f 89 9d d8 1f f8 81 1f 20 
[         0][INFO  : hal_board.c: 601] channel_cnt_table :
[         0][BUF: hal_board.c: 602] 8a a7 e3 a7 3c a8 95 a8 ed a8 46 a9 9f a9 f8 a9 51 aa aa aa 03 ab 5c ab b5 ab 8a ac 94 34 00 23 c4 4e ec 1e cb b7 fc 1e d2 20 0d 1f d8 89 1d 1f df f2 
[         0][BUF: hal_board.c: 602] 2d 1f e5 5b 3e 1f 
[         0][INFO  : hal_board.c: 610] lo_fcal_div : 1387
MAC address mode length 3
MAC address mode is MBF
[         0][INFO  : hal_board.c: 186] Set MAC addrress B4:E8:42:02:9A:5E
[         0][INFO  : hal_board.c: 629] country_code : 86
[         0][ERROR : hal_board.c: 657] pwr_table_11b NULL. lentmp = -1
[         0][ERROR : hal_board.c: 677] pwr_table_11g NULL. lentmp = -1
[         0][ERROR : hal_board.c: 697] pwr_table_11n NULL. lentmp = -1
[        0][INFO  : hal_board.c: 439] pwr_mode length 2
[         0][INFO  : hal_board.c: 443] pwr_mode is bf
[         0][INFO  : hal_board.c: 336] pwr_offset from dtb:
[         0][BUF: hal_board.c: 337] 0a 0a 0a 0a 0a 0a 0a 0a 0a 0a 0a 0a 0a 0a 
[         0][INFO  : hal_board.c: 341] pwr_offset from dtb (rebase on 10):
[         0][INFO  : hal_board.c: 521] ap_ssid string[0] = bl_test_005, ap_ssid_len = 11
[         0][INFO  : hal_board.c: 532] ap_psk string[0] = 12345678, ap_psk_len = 8
[         0][INFO  : hal_board.c: 541] ap_channel = 11
[OS] Starting aos_loop_proc task...
[OS] Starting TCP/IP Stack...
[... 84 lines cut ...]
Init CLI with event Driven
pwm_test 
[     24220][ERROR : bl_pwm.c:  89] arg error. bl_pwm_init freq = 50
[     24220][INFO  : main.c: 210] start pin = 2, id = 2
[     26220][INFO  : main.c: 215] stop pin = 2, id = 2
[     28220][INFO  : main.c: 210] start pin = 2, id = 2
[     30220][INFO  : main.c: 215] stop pin = 2, id = 2
[     32220][INFO  : main.c: 210] start pin = 2, id = 2
[     34220][INFO  : main.c: 215] stop pin = 2, id = 2
[     36220][INFO  : main.c: 210] start pin = 2, id = 2
[     38220][INFO  : main.c: 215] stop pin = 2, id = 2
[     40220][INFO  : main.c: 210] start pin = 2, id = 2
[     42220][INFO  : main.c: 215] stop pin = 2, id = 2
[     44220][INFO  : main.c: 210] start pin = 2, id = 2
[     46220][INFO  : main.c: 215] stop pin = 2, id = 2
[     48220][INFO  : main.c: 210] start pin = 2, id = 2
[     50220][INFO  : main.c: 215] stop pin = 2, id = 2
[     52220][INFO  : main.c: 210] start pin = 2, id = 2
[     54220][INFO  : main.c: 215] stop pin = 2, id = 2
[     56220][INFO  : main.c: 210] start pin = 2, id = 2
[     58220][INFO  : main.c: 215] stop pin = 2, id = 2
[     60220][INFO  : main.c: 210] start pin = 2, id = 2
[     62220][INFO  : main.c: 215] stop pin = 2, id = 2

```

ggggxiaolong, 2020-12-24:
@YafeiJin Could you provide your source code in somewhere that I can down load?

YafeiJin, 2020-12-24:
@ggggxiaolong I tested the patch with the latest code on github, and it works well. Could you please delete build_out and recompile again? Thanks.

YafeiJin, 2020-12-24:
With your branch, it is also correct. So please clean & recompile.

ggggxiaolong, 2020-12-24:
@YafeiJin It works, Thanks

closed 2020-12-27 by YafeiJin