bouffalolab/bl_iot_sdk #28: CKS base register wrong in svd. Overlaps with UART0

btashton, 2020-12-27:
I could not find much information on what the CKS is but the base address defined in the svd overlaps with UART0 at 0x4000a000.  What is the correct base address for this?

shchen-Lab, 2020-12-28:
Hi btashton,

UART0 base address is 0x4000a000. Please refer BL602/604 Datasheet and BL602/604 Reference Manual.

https://dev.bouffalolab.com/media/doc/602/open/datasheet/en/html/content/Functional.html
https://dev.bouffalolab.com/media/doc/602/open/reference_manual/en/html/index.html

btashton, 2020-12-28:
Yes I know.  The base address for CKS is also listed as `0x4000a000` which cannot be correct.

shchen-Lab, 2020-12-28:
Yes, Thanks for your help. CKS base address wrong in svd, we will fix this issue. You can get clock register information at glb_reg.h.
[image]

mkroman, 2021-01-14:
The base address for `CKS` in an updated SVD I got from Bouffalo Lab is `0x4000A700`.

mkroman, 2021-01-18:
I've improved the SVD [here](https://github.com/sipeed/bl602-pac/blob/main/soc602_reg.svd) and I've written some peripheral details [here](https://github.com/pine64/bl602-docs/blob/2251903f8a8eadbbf59299e4fa57acc0bc112b07/hardware_notes/registers/cks.md)