bouffalolab/bl_iot_sdk #33: Bug in static void i2c_transferbytes(i2c_msg_t *pstmsg)?

Kongduino, 2021-01-26:
In [static void i2c_transferbytes(i2c_msg_t *pstmsg)](https://github.com/bouffalolab/bl_iot_sdk/blob/master/components/hal_drv/bl602_hal/hal_i2c.c#L74-L95) there are 2 if conditions that have a single =.

```c
if ((pstmsg->direct == I2C_M_WRITE) && (pstmsg->event = EV_I2C_TXF_INT)) {
```

and

```c
    } else if ((pstmsg->direct == I2C_M_READ) && (pstmsg->event = EV_I2C_RXF_INT)){
```

This looks suspiciously like they should be == equality tests.

lupyuen, 2021-01-26:
The Assignment in the Condition looks suspicious because `pstmsg->event` has already been set by the calling function `i2c_interrupt_entry` ... Yet `pstmsg->event` isn't used after `i2c_transferbytes` returns to `i2c_interrupt_entry`

shchen-Lab, 2021-01-27:
Hi

This may be an issue.  We will check this code, thanks for your information.

shchen-Lab, 2021-01-27:
@lupyuen @Kongduino  This issue will be fixed, thanks!

lupyuen, 2021-01-27:
Awesome thank you :-)

Kongduino, 2021-01-27:
好的👌！