bouffalolab/bl_iot_sdk #35: OTA info

rickbronson, 2021-01-31:
Hi,

  I've searched high and low and can't seem to find any info on how to do an OTA update on the BL602.  I see that tcp port 3333 is used but don't see any applications that use that port to send flash image to the part.  The program BLFlashEnv doesn't seem to know anything about OTA.

 Thanks for any help

  Rick

YafeiJin, 2021-02-01:
@rickbronson If you add bl_sys_ota_cli_init in the project, you will find "ota_tcp" in CLI, and try it. BLFlashEnv can build the OTA bin(bl_iot_sdk/tools/flash_tool/bl602/ota/FW_OTA.bin.ota) by adding OTA header before the compiled bin.  Boot2 also support compressed ota bin(FW_OTA.bin.xz.ota).

rickbronson, 2021-02-01:
HI  YafeiJin,

  Thanks so much for the quick reply.  I seem to already have ota_tcp.  What I was wondering is how do I used it to actually do an OTA update?  I went into BLFlashEnv but didn't see a way to do it from there.  I would prefer a Linux CLI method if at all possible.  Maybe you could show me an example of updating via OTA.

  Is it okay that I use:

image_conf/flash_build.py

  to build the flash OTA images?

  Thanks,

 Rick

YafeiJin, 2021-02-02:
@rickbronson Change make_scripts_riscv/project.mk as the following:

> @cd $(BL60X_SDK_PATH)/image_conf; python3 flash_build.py $(PROJECT_NAME) $(CONFIG_CHIP_NAME)
>     #@cd $(BL60X_SDK_PATH)/image_conf; ./flash_build $(PROJECT_NAME) $(CONFIG_CHIP_NAME)

Change the img_conf/eflash_loader_cfg.conf under the project as the following:

> flash_id = ef4015

Compile the project, then OTA bins will be generated too. FW_OTA.bin.ota and FW_OTA.bin.xz.ota could be used for OTA.
We are preparing one demo of ota over HTTP, which will be better option.
Thanks

YafeiJin, 2021-02-04:
HTTP OTA demo was added

closed 2021-02-04 by YafeiJin