bouffalolab/bl_iot_sdk #43: PIR interrupt example

mikewen, 2021-02-19:
How to use PIR interrupt?

Which pins can use for PIR detection?

shchen-Lab, 2021-02-19:
ir rx gpio (gpio11~gpio13) and ir output only gpio11 suooprt. and ir support led chip include WS2812B and UCS1903.
Please add below code to your deivce tree. 
{
        #address-cells = <1>;
        #size-cells = <1>;
        ctrltype = <0>;
        tx {
            status = "disable";
            pin = <11>;         // only support 11
            mode = "NEC";       // NEC、ExtenedNEC、RC5、SWM
            interval = <100>;   // ms
            active_mode = "Hi"; //Hi、Lo
        };
        rx {
            status = "okay";
            pin = <12>;         // only support 12 13
            mode = "NEC";       // NEC、ExtenedNEC、RC5、SWM
            active_mode = "Hi"; //Hi、Lo
        };
    };
You can find below code at sdk_app_ir :
``static void event_cb_ir_event(input_event_t *event, void *private_data)
{
    switch (event->code) {
        case CODE_IR_RECEIVE_NEC:
        {
            printf("[IR] [EVT] RECEIVE DATA @%llums, RAW %08lx, ADDR %02x, CMD %02x\r\n", aos_now_ms(), (unsigned long)event->value,
                    bl_ir_get_addr((unsigned int)event->value),
                    bl_ir_get_cmd((unsigned int)event->value)
            );
        }
        break;
        case CODE_IR_RECEIVE_NEC_REPEAT:
        {
            printf("[NEC] Repeat @%llums\r\n", aos_now_ms());
        }
        break;
        default:
        {
            printf("[IR] [EVT] Unknown code %u, %lld\r\n", event->code, aos_now_ms());
            /*nothing*/
        }
   }
}

gamelaster, 2021-02-19:
Hi @mikewen , there is huge chance that the reply from developers will be delayed, due they celebrating Chinese New Year.

YafeiJin, 2021-02-19:
@mikewen If you want to wake up BL602 by one GPIO which is connected to one external PIR, please add the following into bl_factory_params_IoTKitA_40M.dts

> 

        gpio7 {
            status = "okay";
            pin = <7>;
            feature = "button";
            active = "Hi";
            mode = "multipress";
            button {
                debounce = <10>;
                short_press_ms {
                    start = <100>;
                    end = <3000>;
                    kevent = <2>;
                };
                long_press_ms {
                    start = <6000>;
                    end = <10000>;
                    kevent = <3>;
                };
                longlong_press_ms {
                    start = <15000>;
                    kevent = <4>;
                };
                trig_level = "Hi";
            };
            hbn_use = "okay";
        };

Then call `hal_hbn_enter` in code.
Please double check the GPIO7 is not PULL-UP with your board, cause GPIO7 is used as UART RX by default. Maybe you need some rework. Thanks

mikewen, 2021-02-19:
@YafeiJin  Thank you!

I guess use hal_hbn_enter  to enter the sleep mode? (stand for 'hal hibernate enter' ?)