bouffalolab/bl_iot_sdk #54: How do I get BL602_BOOT2_MINI to work and print the log?

balanceTWK, 2021-05-26:
[image]

[image]

gamelaster, 2021-05-26:
The bl602_boot2_mini itself is the "Boot 2 Bin", so you need to set this .bin as Boot 2 Bin.

balanceTWK, 2021-05-26:
> The bl602_boot2_mini itself is the "Boot 2 Bin", so you need to set this .bin as Boot 2 Bin.

Yes, I tried downloading the program like this. But there is no reaction and I am not sure if my BL602_BOOT2_MINI is running correctly.

shchen-Lab, 2021-06-02:
@balanceTWK  

bl602_boot2_mini  is boot program,you need to set this .bin as Boot 2 Bin.  bl602_boot2_mini can't set Firmware Bin. you need set bl602_demo_event bin as Firmware Bin.