bouffalolab/bl_iot_sdk #55: Issues with C++

JF002, 2021-07-03:
Hi!

I'm trying to port a C++ application to the BL602 using  this SDK. This app use some parts of the STL like std::unique_ptr and std::array, for example.
All my attempts to build and run my app failed, either at build/link time or at runtime.

I tried to build a much simpler application based on sdk_app_helloworld. Here's a first example that actually works:

MyCxxClass.h
```
#ifndef __MYCXXCLASS__
#define __MYCXXCLASS__

#include <stdio.h>
class MyCxxClass {
public:
  MyCxxClass() = default;
  ~MyCxxClass() = default;
  void Test() {
    printf("Hi!\n");
  }
};

#endif
```

App.cpp:
```
#include "MyCxxClass.h"

extern "C" {
void RunMyApp(void) {
  MyCxxClass c;
  c.Test();
}
}
```

main.cpp:
```
void RunMyApp(void);

void bfl_main(void)
{
    /*
     * Init UART using pins 16+7 (TX+RX)
     * and baudrate of 2M
     */
    bl_uart_init(0, 16, 7, 255, 255, 2 * 1000 * 1000);    
    helloworld();
    RunMyApp();

}
```

Now, let's try with a slightly modified App.cpp:
```
#include "MyCxxClass.h"
#include <memory>

std::unique_ptr<MyCxxClass> c;

extern "C" {
void RunMyApp(void) {
  c->Test();
}
}
```

This time, the link fails:

```
$ make -j
use exsting version.mk file
CXX build_out/sdk_app_helloworld/App.o
AR build_out/sdk_app_helloworld/libsdk_app_helloworld.a
LD build_out/sdk_app_helloworld.elf
/home/jf/git/test/bl_iot_sdk/toolchain/riscv/Linux/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/jf/git/test/bl_iot_sdk/customer_app/sdk_app_helloworld/build_out/sdk_app_helloworld/libsdk_app_helloworld.a(App.o): in function `_GLOBAL__sub_I_c':
/home/jf/git/test/bl_iot_sdk/customer_app/sdk_app_helloworld/sdk_app_helloworld/App.cpp:4: undefined reference to `__dso_handle'
/home/jf/git/test/bl_iot_sdk/toolchain/riscv/Linux/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/jf/git/test/bl_iot_sdk/customer_app/sdk_app_helloworld/sdk_app_helloworld/App.cpp:4: undefined reference to `__dso_handle'
/home/jf/git/test/bl_iot_sdk/toolchain/riscv/Linux/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/jf/git/test/bl_iot_sdk/customer_app/sdk_app_helloworld/build_out/freertos_riscv_ram/libfreertos_riscv_ram.a(heap_5.o): in function `vPortFree':
/home/jf/git/test/bl_iot_sdk/components/bl602/freertos_riscv_ram/portable/MemMang/heap_5.c:300: undefined reference to `vAssertCalled'
/home/jf/git/test/bl_iot_sdk/toolchain/riscv/Linux/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/jf/git/test/bl_iot_sdk/components/bl602/freertos_riscv_ram/portable/MemMang/heap_5.c:329: undefined reference to `vAssertCalled'
/home/jf/git/test/bl_iot_sdk/toolchain/riscv/Linux/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/jf/git/test/bl_iot_sdk/customer_app/sdk_app_helloworld/build_out/freertos_riscv_ram/libfreertos_riscv_ram.a(tasks.o): in function `xTaskIncrementTick':
/home/jf/git/test/bl_iot_sdk/components/bl602/freertos_riscv_ram/tasks.c:2843: undefined reference to `vAssertCalled'
/home/jf/git/test/bl_iot_sdk/toolchain/riscv/Linux/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/jf/git/test/bl_iot_sdk/customer_app/sdk_app_helloworld/build_out/freertos_riscv_ram/libfreertos_riscv_ram.a(tasks.o): in function `xTaskResumeAll':
/home/jf/git/test/bl_iot_sdk/components/bl602/freertos_riscv_ram/tasks.c:2254: undefined reference to `vAssertCalled'
/home/jf/git/test/bl_iot_sdk/toolchain/riscv/Linux/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/jf/git/test/bl_iot_sdk/customer_app/sdk_app_helloworld/build_out/sdk_app_helloworld.elf: hidden symbol `__dso_handle' isn't defined
/home/jf/git/test/bl_iot_sdk/toolchain/riscv/Linux/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make: *** [/home/jf/git/test/bl_iot_sdk/make_scripts_riscv/project.mk:439: /home/jf/git/test/bl_iot_sdk/customer_app/sdk_app_helloworld/build_out/sdk_app_helloworld.elf] Error 1
```

I can of course define `vAssertCalled` but I still have this issue with `__dso_handle` which is, if I understand correctly, a part of the C++ stdlib.

Is there anything I can do to build a C++ application based on this SDK?

Thanks!

shchen-Lab, 2021-07-10:
Please test below code, this code work well on my platform.
`#include <stdio.h>
#include <string.h>

class Box
{
   public:
      double length;   
      double breadth; 
      double height;   
     
      double get(void);
      void set( double len, double bre, double hei );
};

double Box::get(void)
{
    return length * breadth * height;
}
 
void Box::set( double len, double bre, double hei)
{
    length = len;
    breadth = bre;
    height = hei;
}

extern "C" void Test_box(void)
{
    Box Box1;
    double volume = 0.0;     
    Box1.height = 5.0; 
    Box1.length = 6.0; 
    Box1.breadth = 7.0;
    volume = Box1.get(); 
    printf("Box1 get=%f\n",volume);

}`

shchen-Lab, 2021-07-10:
Please test attacthment, this code work well on my platform.
[sdk_app_helloworld.zip](https://github.com/bouffalolab/bl_iot_sdk/files/6794567/sdk_app_helloworld.zip)

robertlipe, 2021-07-14:
For the dso_handle issue, just make a global:
void* __dso_handle = nullptr;
For related issues, see the (excellent) https://alex-robenko.gitbook.io/bare_metal_cpp/compiler_output/static and related chapters.