bouffalolab/bl_iot_sdk #59: 请教下，我使用最新得SDK，版本为1.6.31和1.6.32。编译你们得demo编译不过，我以前使用的是1.6.22得版本，编译是OK的。我在windows10进行编译

LzhBee, 2021-12-04:
Hzzz@LAPTOP-7EDE7524 MSYS /d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/customer_app/bl602_demo_event
$ ./genromap
****** Please SET BL60X_SDK_PATH ******
****** Trying SDK PATH [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/customer_app/bl602_demo_event/../..]
use existing version.mk file
****** Please SET BL60X_SDK_PATH ******
****** Trying SDK PATH [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/customer_app/bl602_demo_event/../..]
use existing version.mk file
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/external_libs/jsmn/jsmn.o
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_jobs_interface.o
AS build_out/bl602/evb/src/boot/gcc/entry.o
CC build_out/bl602_demo_event/aes_test.o
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_jobs_json.o
AS build_out/bl602/evb/src/boot/gcc/start.o
CC build_out/bl602_demo_event/aws_iot_main.o
CC build_out/bl602_demo_event/ble.o
CC build_out/bl602/evb/src/debug.o
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_jobs_topics.o
CC build_out/bl602_demo_event/blsync_ble_app.o
CC build_out/bl602/evb/src/strntoumax.o
CC build_out/bl602_demo_event/coex_debug.o
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_jobs_types.o
CC build_out/bl602/bfl_main.o
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_json_utils.o
CC build_out/blcrypto_suite/src/blcrypto_suite_bignum.o
CC build_out/bl602_demo_event/main.o
CC build_out/bl602_demo_event/rijndael.o
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client.o
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_common_internal.o
CC build_out/blcrypto_suite/src/blcrypto_suite_ecp.o
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_connect.o
CC build_out/bl602_demo_event/wifi_prov_api.o
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_publish.o
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：aes_test.o] 错误 1
make[1]: *** 正在等待未完成的任务....
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：aws_iot_main.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：evb/src/boot/gcc/start.o] 错误 1
make[1]: *** 正在等待未完成的任务....
CC build_out/blcrypto_suite/src/blcrypto_suite_ecp_curves.o
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_subscribe.o
CC build_out/blcrypto_suite/src/blcrypto_suite_platform_util.o
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:312：aws-iot-device-sdk-embedded-C/src/aws_iot_jobs_interface.o] 错误 1
make[1]: *** 正在等待未完成的任务....
CC build_out/aws-iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_unsubscribe.o
CC build_out/blcrypto_suite/src/blcrypto_suite_porting.o
CC build_out/blmtd/bl_mtd.o
CC build_out/blfdt/src/fdt.o
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：aws-iot-device-sdk-embedded-C/external_libs/jsmn/jsmn.o] 错误 1
CC build_out/blfdt/src/fdt_ro.o
CC build_out/blestack/src/port/bl_port.o
CC build_out/blog/blog.o
CC build_out/bl602_std/bl602_std/StdDriver/Src/bl602_uart.o
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：ble.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：aws-iot-device-sdk-embedded-C/src/aws_iot_jobs_topics.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：evb/src/strntoumax.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：aws-iot-device-sdk-embedded-C/src/aws_iot_jobs_types.o] 错误 1
CC build_out/blfdt/src/fdt_wip.o
CC build_out/blestack/src/common/atomic_c.o
CC build_out/bl602_std/bl602_std/StdDriver/Src/bl602_adc.o
CC build_out/blcrypto_suite/src/blcrypto_suite_hacc.o
CC build_out/blcrypto_suite/src/blcrypto_suite_aes.o
CC build_out/bl602_std/bl602_std/StdDriver/Src/bl602_sec_eng.o
CC build_out/blestack/src/common/buf.o
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：evb/src/boot/gcc/entry.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：aws-iot-device-sdk-embedded-C/src/aws_iot_jobs_json.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：blsync_ble_app.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bfl_main.o] 错误 1
CC build_out/blog_testc/blog_testc.o
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：rijndael.o] 错误 1
CC build_out/blog_testc/blog_testc1_diable.o
[... 216 lines cut ...]
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bl602_std/StdDriver/Src/bl602_spi.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/host/keys.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bl602_std/Common/soft_crc/softcrc.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/host/settings.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/common/tinycrypt/source/ctr_mode.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bl602_std/StdDriver/Src/bl602_mfg_efuse.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/cli_cmds/pts_cli_cmds.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/common/tinycrypt/source/ecc_dh.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bl602_std/StdDriver/Src/bl602_mfg_flash.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/services/ble_tp_svc.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bl602_std/StdDriver/Src/bl602_sf_cfg_ext.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/host/uuid.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/cli_cmds/ble_cli_cmds.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bl602_std/StdDriver/Src/bl602_sflash_ext.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bl602_std/StdDriver/Src/bl602_xip_sflash_ext.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/host/bl_host_assist.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/host/conn.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/cli_cmds/bredr_cli_cmds.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bl602_std/Common/xz/xz_port.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/host/gatt.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bl602_std/Common/platform_print/platform_gpio.o] 错误 1
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：src/bl_hci_wrapper/bl_hci_wrapper.o] 错误 1
make: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/customer_app/bl602_demo_event/../../make_scripts_riscv/project.mk:563：component-blestack-build] 错误 2
make[1]: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/make_scripts_riscv/component_wrapper.mk:313：bl602_std/Common/cipher_suite/src/bflb_ecdsa.o] 错误 1
make: *** [/d/bl_iot_sdk-master-1.6.32/bl_iot_sdk-master/customer_app/bl602_demo_event/../../make_scripts_riscv/project.mk:563：component-bl602_std-build] 错误 2

LzhBee, 2021-12-04:
linux下编译最新的SDK编译过了。

cmheia, 2021-12-08:
工具链的问题。别问，毕竟是盗的。

shchen-Lab, 2022-01-08:
保证工具链10.2更新一下工具链，看起来是工具链的问题，你可以参考我的配置如下图：
[image]
[image]

shchen-Lab, 2022-01-08:
MSYS2的工具链问题：
[image]

设置一下环境变量就ok

closed 2022-01-08 by shchen-Lab