bouffalolab/bl_iot_sdk #60: How to build Zigbee sample firmware?

tomasz-s, 2022-01-13:
I bought some XT-ZB1 DevKit boards and want to run some Zigbee on it.
Basic samples compiles ok but Zigbee sample cannot find components.
but when I trying "./genzsed" compilation failed
 
 /root/bl_iot_sdk-master/components/network/zigbee/zigbee_def.mk: No such file or directory

shchen-Lab, 2022-01-21:
> 
BL702 zigbee SDK is only avaliable when NDA is signed. Please contact our sales team.

closed 2022-05-17 by tomasz-s

turboJo, 2022-05-17:
这是来自QQ邮箱的假期自动回复邮件。您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

joshOberhaus, 2022-09-21:
> I bought some XT-ZB1 DevKit boards and want to run some Zigbee on it. Basic samples compiles ok but Zigbee sample cannot find components. but when I trying "./genzsed" compilation failed
> 
> /root/bl_iot_sdk-master/components/network/zigbee/zigbee_def.mk: No such file or directory

Did you end up getting access? I've been requesting the NDA for weeks.