bouffalolab/bl_iot_sdk #78: BL602 - C/L/E - What's different between these?

divadiow, 2024-02-28:
OK, so the datasheet says combo/light/enhance, but what's the ACTUAL difference between them? what does light mean? what does enhance mean?

thanks
[image]

gamelaster, 2024-02-29:
Hi @divadiow 
Those repositories are not maintained much, and generally, I don't think you will get answer.
I remember some datasheets had some tiny description about those, but not much. But take look on list of packages available for purchase, and some options might not have all RF features available.

divadiow, 2024-03-02:
Sure. I can only find reference to what I posted, no details. Oh well