bouffalolab/bl_iot_sdk #81: mac环境下总是提示“Please Press Reset Key!”

mindops, 2024-08-04:
使用的安信可的模块，在mac下，总是提示“Please Press Reset Key!”，在windows环境是可以的，请问是什么原因，是不兼容吗

[21:38:46.628] - ========= load eflash_loader.bin =========
[21:38:46.628] - Load eflash_loader.bin via uart
[21:38:46.628] - ========= image load =========
[21:38:46.642] - Please Press Reset Key!
[21:39:49.067] - default set DTR high
[21:39:49.174] - clean buf
[21:39:49.179] - send sync
[21:39:49.285] - ack is b''
[21:39:49.288] - retry
[21:39:49.290] - default set DTR high
[21:39:49.398] - clean buf
[21:39:49.405] - send sync
[21:39:49.513] - ack is b''
[21:39:49.517] - retry
[21:39:49.522] - shake hand fail
[21:39:49.524] - ########################################################################
[21:39:49.526] - 请按照以下描述排查问题：
[21:39:49.528] - GPIO8是否上拉到板子自身的3.3V，而不是外部的3.3V
[21:39:49.529] - GPIO7(RX)是否连接到USB转串口的TX引脚
[21:39:49.531] - GPIO16(TX)是否连接到USB转串口的RX引脚
[21:39:49.533] - 在使用烧录软件进行烧录前，是否在GPIO8拉高的情况下，使用Reset/Chip_En复位了芯片
[21:39:49.534] - 烧录软件所选择的COM口，是否是连接芯片的串口
[21:39:49.536] - 烧录软件上选择的波特率是否是USB转串口支持的波特率
[21:39:49.537] - 3.3V供电是否正常
[21:39:49.539] - 板子供电电流是否正常(烧录模式下，芯片耗电电流5-7mA)
[21:39:49.541] - ########################################################################
[21:39:49.543] - shake hand fail
[21:39:49.550] - Load helper bin time cost(ms): 62921.478759765625
[21:39:49.551] - {"ErrorCode": "0050","ErrorMsg":"BFLB IMG LOAD SHAKEHAND FAIL"}
[21:39:49.553] - {"ErrorCode": "0003","ErrorMsg":"BFLB LOAD HELP BIN FAIL"}
[21:39:49.554] - Burn Retry
[21:39:49.556] - 0
[21:39:49.557] - Burn return with retry fail

gamelaster, 2024-08-04:
macOS together with FreeBSD have very problematic UART driver.
Try to use baudrate 115200, but DevCube changes this baudrate in eflash_loader, causing troubles.
You can try blisp, but only few formats are supported: https://github.com/pine64/blisp

---

macOS 和 FreeBSD 的 UART 驱动程序存在很大问题。
您可以尝试使用波特率 115200，但 DevCube 会在 eflash_loader 中更改波特率，从而导致问题。
您可以尝试使用 blisp，但它只支持以下几种格式： https://github.com/pine64/blisp

mindops, 2024-08-06:
Thank you for your reply, but I'm still getting an error when I execute the command using blisp：
```
./blisp write --chip bl60x --reset -p /dev/tty.usbserial-2110 ~/soft/Ai-Thinker-WB2/applications/get-started/helloworld/build_out/helloworld.bin
Sending a handshake...
Received no response from chip.
Failed to handshake with device.
```
I am compiling and running on the Mac M1 platform.