bouffalolab/bl_iot_sdk #86: bl_iot_sdk的串口驱动是否支持UART_URX_RTO_TIMER？

FlyyingPiggy2020, 2024-10-27:
阅读寄存器知道了有一个UART_URX_RTO_TIMER，串口接收超时的中断。

按我的理解应该是超出时间之后会产生一次中断。

然后阅读了一下串口驱动的代码，我好像没有看到它把中断的触发源提供给回调函数。[（看这里）](https://github.com/bouffalolab/bl_iot_sdk/blob/edcf613f073d3ef7d2f290d7c285b69be35da75b/components/platform/hosal/bl602_hal/bl_uart.c#L424-L436)

对于我上层的应用来说，触发`UART_URX_FIFO_INT`和`UART_URX_RTO_INT`是分不清楚的。

或者说这个超时中断使用有别的姿势和用法？
请问当初设计的时候，这个地方应该是如何使用的？

FlyyingPiggy2020, 2024-10-27:
我的想法是这样的，
可以给超时中断单独设置一个回调函数。

或者把`bl_uart.c`426行的`BL_WR_REG(UARTx,UART_INT_CLEAR,0x10);`这句话放到回调函数后面？

这样子的话，在外面也可以去用`BL_IS_REG_BIT_SET(tmpVal,UART_URX_RTO_INT)`查询中断是否触发了。

gamelaster, 2024-10-27:
Check out also Bouffalo SDK, the UART drivers for BL702 should be almost the same for BL602, and they're newer, so take look on them.

FlyyingPiggy2020, 2024-10-28:
@gamelaster Will Bouffalo SDK have plans to support bl602/bl604 in the future?I noticed that LHAL currently supported bl602, but it hasn't been tested, and wireless functionality is not supported.

gamelaster, 2024-10-28:
@FlyyingPiggy2020 I don't think so. The drivers should work, but I don't think we will see WiFi support in it.